osrf/ros2_test_cases #752: Local workspace overlay

Yadunund, 2023-04-05:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Ubuntu Jammy

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
I'll take this one.

tylerjw, 2023-05-03:
Tested this and it works as expected. Did the same test as the other issue except this time depending on the binary tarbal.

Underlay:
```
$ vcs export
repositories:
  RSL:
    type: git
    url: git@github.com:PickNikRobotics/RSL.git
    version: main
  cpp_polyfills:
    type: git
    url: git@github.com:PickNikRobotics/cpp_polyfills.git
    version: main
```

Overlay (`generate_parameter_library` depends on `RSL`):
```
repositories:
  generate_parameter_library:
    type: git
    url: git@github.com:PickNikRobotics/generate_parameter_library.git
    version: main
```

Steps tested:
1. Build underlay based on iron (`colcon build`)
2. Run tests in underlay (`colcon test`)
3. Source underlay workspace (`source install/setup.bash`)
4. Build overlay workspace
5. Test overlay workspace

OS: Ubuntu Jammy 22.04
DDS: Cyclone

closed 2023-05-03 by clalancette