osrf/ros2_test_cases #754: Local workspace overlay

Yadunund, 2023-04-05:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Arm64
- Os: Ubuntu Jammy

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-06:
**Setup**

DDS vendor: Cyclone
BuildType: Binary
Chip: Arm64
Os: Ubuntu Jammy

A package which depends on rclcpp and sensor_msgs is built successfully. And the node is running as expected. The check is pass.

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette