osrf/ros2_test_cases #763: Local workspace overlay

Yadunund, 2023-04-05:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-06:
**Setup**

- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- OS: Rhel 9

I built a package which depends on rclcpp and sensor_msgs resulting in successful build. This check is passed.

clalancette, 2023-05-07:
Thank you for testing!

closed 2023-05-07 by clalancette