osrf/ros2_test_cases #771: Parameter executables in `demo_nodes_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Async parameter client**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py async_param_client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-07:
Hello, could you please assign this issue to me? Thank you so much!

juliamp22, 2023-05-07:
**Async parameter client** - Works fine:
[image]

juliamp22, 2023-05-07:
This Issue can be closed, as it works fine! Thank you so much.

clalancette, 2023-05-07:
Thank you for testing!

closed 2023-05-07 by clalancette