osrf/ros2_test_cases #773: Parameter executables in `demo_nodes_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Async parameter client**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py async_param_client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-05:
On a Windows 11 binary installation.

I have run the commands, and seen everything successful on the async_param_client side, no error reported.

[image]

Yadunund, 2023-05-05:
Thanks for testing!

closed 2023-05-05 by Yadunund