osrf/ros2_test_cases #774: Parameter executables in `demo_nodes_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Async parameter client**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py async_param_client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-06:
Hey, I would like to work on this ticket.

Kartik9250, 2023-05-06:
[image]

Both the commands work. Not sure what i should be expecting or looking for here though.

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette