osrf/ros2_test_cases #783: Quality of Service Demos

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Demo URL](https://github.com/ros2/demos/blob/master/quality_of_service_demo/README.md)

## Checks
- [ ] **Deadline with pause**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 2000
       ```
  
    **Expect**
    
    1.
       The publisher will publish for 5 seconds, then pause for 2 - deadline events will start firing on both participants, until the publisher comes back.
    
  </details>
- [ ] **Deadline no pause**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 0
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 0
       ```
  
    **Expect**
    
    1.
       The publisher doesn't actually pause, no deadline misses should occur.
    
  </details>
- [ ] **Lifespan, some survive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan 1000 --publish-count 10 --subscribe-after 3000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan 1000 --publish-count 10 --subscribe-after 3000
       ```
  
    **Expect**
    
    1.
       After a few seconds, you should see (approximately) messages 4-9 printed from the subscriber.
       
       The first few messages, with 1 second lifespan, were gone by the time the subscriber joined after 3 seconds.
       
    
  </details>
- [ ] **Lifespan, all survive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan 4000 --publish-count 10 --subscribe-after 3000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan 4000 --publish-count 10 --subscribe-after 3000
       ```
  
    **Expect**
    
    1.
       After a few seconds, you should see all of the messages (0-9) printed from the subscriber.
       
       All messages, with their 4 second lifespan, survived until the subscriber joined.
       
    
  </details>
- [ ] **Liveliness**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py liveliness 1000 --kill-publisher-after 2000
       ```
  
    **Expect**
    
    1.
       After 2 seconds, the publisher will be killed, and the subscriber will receive a callback 1 second after that notifying it that the liveliness has changed.
    
  </details>
- [ ] **Incompatible QoS Offered/Requested**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp incompatible_qos durability
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py incompatible_qos durability
       ```
  
    **Expect**
    
    1.
       The publisher in this demo will publish 5 messages, but also output a notification stating that its offered QoS is incompatible with the subscriber that is also created by this demo. The subscriber in this demo will similarly output a notification stating that its requested QoS is incompatible with the QoS offered by the publisher.
       
    
  </details>
- [ ] **Interactive Quality of Service Demo**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp interactive_publisher
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run quality_of_service_demo_cpp interactive_subscriber
       ```
    1.
       In the publisher's terminal, press `p` to manually assert the liveliness of the publisher, for when Liveliness kind is `MANUAL_BY_TOPIC`.
       
    1.
       In the publisher's terminal, press `s` to toggle enabling/disabling of publishing messages
       
    1.
       In the publisher and subscriber's terminal, press `q` to print the current QoS policy.
       
    1.
       In the publisher and subscriptions's terminal, press `x` to stop and exit the demo.
       
  
    **Expect**
    
    1.
       All key-press commands work.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-06:
### Checks

- [x] Deadline with pause
- Running the command below gives this warning but the script runs. 
>   2. ```shell
>      # User input in terminal 1
>      ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 2000
>      ```
warning: `/opt/ros/iron/lib/python3.10/site-packages/rclpy/qos_event.py:19: UserWarning: importing 'qos_event' is deprecated; import 'event_handler' instead
  warnings.warn("importing 'qos_event' is deprecated; import 'event_handler' instead")`
- In the `Python version` it gives out this when the dealine event starts:
`[INFO] [1683354384.494594782] [Listener]: <rclpy._rclpy_pybind11.rmw_requested_deadline_missed_status_t object at 0x7fea96ffef30>
[INFO] [1683354384.495484924] [Talker]: <rclpy._rclpy_pybind11.rmw_offered_deadline_missed_status_t object at 0x7fea96ffff30>`
- While in the `C version` it gives out:
`[INFO] [1683354570.389331166] [talker]: Offered deadline missed - total 1 delta 1
[INFO] [1683354570.389493227] [listener]: Requested deadline missed - total 1 delta 1`
- Also when I run the command, talker-listener communicates till 6 seconds at first then after the deadline event starts(2sec) that comes down to 3 sec.

<br>

- [x] Deadline with no pause
- While running the `C version` there is a delay after the deadline event was supposed to start due to which the listener heard only after talker published 10.(not on the terminal)
`[INFO] [1683355016.929476701] [talker]: Publishing: 'Talker says 9'
[INFO] [1683355016.929801428] [talker]: Publishing: 'Talker says 10'
[INFO] [1683355016.929934771] [listener]: Listener heard: [Talker says 9]
[INFO] [1683355016.929975669] [listener]: Listener heard: [Talker says 10]`
- The `python ver` works just fine, there;s just one warning that i mentioned above.

<br>

- [x] Lifespan, some survive
- In the `C version` similar delay as I mention above was seen.
`[INFO] [1683355766.728818198] [talker]: Publishing: 'Talker says 4'
[INFO] [1683355767.228743285] [talker]: Publishing: 'Talker says 5'
[INFO] [1683355767.229830913] [listener]: Listener heard: [Talker says 4]
[INFO] [1683355767.229885045] [listener]: Listener heard: [Talker says 5]
[INFO] [1683355767.728800890] [talker]: Publishing: 'Talker says 6'
[INFO] [1683355767.729062856] [listener]: Listener heard: [Talker says 6]
`
- In the `Python version` similar delay to the `C version` was seen but instead of just Talker straightaway publishing 5, 'subscription created was heard'
`[INFO] [1683355893.208808146] [qos_talker]: Publishing: Talker says 4
[INFO] [1683355893.710443030] [qos_listener]: Subscription created
[INFO] [1683355893.711323536] [qos_talker]: Publishing: Talker says 5
[INFO] [1683355893.712304171] [qos_listener]: I heard: Talker says 4
[INFO] [1683355893.713397069] [qos_listener]: I heard: Talker says 5
[INFO] [1683355894.209293065] [qos_talker]: Publishing: Talker says 6
[INFO] [1683355894.210588986] [qos_listener]: I heard: Talker says 6
`
- Also in `Python version` exiting the script by using `Ctrl+C` gives out this error before successfully stopping and exiting the script.
`^CTraceback (most recent call last):
  File "/opt/ros/iron/lib/quality_of_service_demo_py/lifespan", line 33, in <module>
    sys.exit(load_entry_point('quality-of-service-demo-py==0.27.0', 'console_scripts', 'lifespan')())
  File "/opt/ros/iron/lib/python3.10/site-packages/quality_of_service_demo_py/lifespan.py", line 72, in main
    executor.spin()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 280, in spin
    self.spin_once()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 717, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 703, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 600, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt`

- Both the script stops after listener heard 9, that is an intended function i think.

<br>

- [x]  Lifespan, all survive
[... 56 lines cut ...]
- [x]  Interactive Quality of Service Demo
- `p` key throws out this error:
`asserting publisher liveliness
2023-05-06 13:08:56.736 [DATAWRITER Error] Could not assert liveliness of writer 01.0f.0a.33.16.1e.5a.62.00.00.00.00|0.0.12.3 -> Function assert_liveliness`
- `s` key works and stops sending messages, starts sending messages
- `q` key prints current QoS Policy, works on both publisher's and subscriber's terminal.
- `x` key prints out that it is `exiting the demo` But, it does not. `Ctrl+C` does not works have to kill the process with `Ctrl+X` 

<br>

### Few things to point out

- One difference that i noted was that in `C version` it notifies us that both the talker and listener and are starting up at the begining whereas `python version` only notifies us about talker starting up at the start.
- And just before `python version` starts to listen it then notifies us that subscription was created.
- Both version have a different formatting, I wonder if they could be unified. Found `C version` to be more informative and easier to read.
- Python outputs `Object classes` at some places.(Mention above)
- Python gives out error while running the script in the following cases, also mentioned above:
    - Deadline with pause
    - Deadline no pause
    - Liveliness
    -  Incompatible QoS Offered/Requested
- Python gives out error while exiting the script in the following cases, also mentioned above:
    - Lifespan, some survive
    - Lifespan, all survive
    - Liveliness

Kartik9250, 2023-05-06:
Sorry, I forgot to ask you to assign me this ticket before starting working on this.

clalancette, 2023-05-08:
> /opt/ros/iron/lib/python3.10/site-packages/rclpy/qos_event.py:19: UserWarning: importing 'qos_event' is deprecated; import 'event_handler' instead
warnings.warn("importing 'qos_event' is deprecated; import 'event_handler' instead")

See https://github.com/ros2/demos/pull/615, which fixes this particular issue.

clalancette, 2023-05-11:
> * In the `Python version` it gives out this when the dealine event starts:
> 
> 
> ```shell
> [INFO] [1683354384.494594782] [Listener]: <rclpy._rclpy_pybind11.rmw_requested_deadline_missed_status_t object at 0x7fea96ffef30>
> [INFO] [1683354384.495484924] [Talker]: <rclpy._rclpy_pybind11.rmw_offered_deadline_missed_status_t object at 0x7fea96ffff30>
> ```

Yes, good feedback.  See https://github.com/ros2/demos/pull/626 which fixes the Python version to print the same things as the C++ one.

Kartik9250, 2023-05-13:
@clalancette hey, how do i check these fixes.

clalancette, 2023-05-17:
> @clalancette hey, how do i check these fixes.

We've backported *some* of the fixes to Iron, but not all of them.  So you may not see these fixed in Iron right now.

The best way to check on the fixes is to build Rolling from source: http://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html .  From there you should be able to run the demos and see the latest and greatest fixes.

Yadunund, 2023-05-18:
Closing since the fixes have been merged in.

closed 2023-05-18 by Yadunund

clalancette, 2023-05-19:
Just as an FYI, I've also opened up https://github.com/ros2/demos/pull/632 to fix a few more issues in here, and https://github.com/ros2/demos/issues/633 to track the final issue (since it is complex enough that I won't have time for it in the near future).

Thanks for your testing here; we made a lot of improvements to these demos based on it.