osrf/ros2_test_cases #790: ROS 1 bridge

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/ros1_bridge)

## Checks
- [ ] **Build ros1_bridge**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the README to compile the bridge
       
  
    
  </details>
- [ ] **Run the birdge with talker and listener**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 1.
       
  
    
  </details>
- [ ] **Run the birdge and exchange images**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 2.
       
  
    
  </details>
- [ ] **Run the birdge and AddTwoInts service**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 3.
       
  
    
  </details>
- [ ] **Run the bridge with only selected topics and services.**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 4.
       
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-13:
Hello @Yadunund do I have to install any ROS1 version for this Issue? 

I have downloaded the package and tried to build it, and it gives me the following error:

[image]

Yadunund, 2023-05-18:
Closing this one out since ROS 1 binaries are not distributed on Windows.

closed 2023-05-18 by Yadunund

clalancette, 2023-05-18:
> @clalancette is this test case valid? Since we expect users to install ROS binaries from upstream Ubuntu, I'm assuming windows users do not have a way to test this?

Yeah, that's correct.  We should remove this one from YATM.