osrf/ros2_test_cases #792: ROS 1 bridge

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/ros1_bridge)

## Checks
- [ ] **Build ros1_bridge**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the README to compile the bridge
       
  
    
  </details>
- [ ] **Run the birdge with talker and listener**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 1.
       
  
    
  </details>
- [ ] **Run the birdge and exchange images**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 2.
       
  
    
  </details>
- [ ] **Run the birdge and AddTwoInts service**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 3.
       
  
    
  </details>
- [ ] **Run the bridge with only selected topics and services.**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 4.
       
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
Gonna give this a go

bai-ivan, 2023-05-07:
Failed after a long compile. Using the default instructions provided. Can't find google results on the issue. If someone has hints to it - i can give it a go again.
```
--- stderr: ros1_bridge
CMake Warning (dev) at cmake/find_ros1_interface_packages.cmake:53 (set):
  Only the first value argument is used when setting an environment variable.
  Argument 'C:\Users\User\bridge\install\stereo_msgs\Lib\site-packages'
  and later are unused.
Call Stack (most recent call first):
  CMakeLists.txt:53 (find_ros1_interface_packages)
This warning is for project developers.  Use -Wno-dev to suppress it.

ERRORFailed to load RTI Connext DDS Micro
---
Failed   <<< ros1_bridge [44min 44s, exited with code 1]

Summary: 0 packages finished [44min 44s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

Voldivh, 2023-06-02:
Taking into account this [discussion](https://github.com/osrf/ros2_test_cases/issues/790#issuecomment-1552753656) shouldn't we close this one as well?

bai-ivan, 2023-06-02:
Yea, you can close it :)

closed 2023-06-06 by Yadunund