osrf/ros2_test_cases #793: ROS 1 bridge

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/ros1_bridge)

## Checks
- [ ] **Build ros1_bridge**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the README to compile the bridge
       
  
    
  </details>
- [ ] **Run the birdge with talker and listener**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 1.
       
  
    
  </details>
- [ ] **Run the birdge and exchange images**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 2.
       
  
    
  </details>
- [ ] **Run the birdge and AddTwoInts service**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 3.
       
  
    
  </details>
- [ ] **Run the bridge with only selected topics and services.**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 4.
       
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fmrico, 2023-05-14:
Hi @Yadunund 

I would like to contribute to this issue. Please, assign it to me.

Thanks

fmrico, 2023-05-15:
- [X] **Build ros1_bridge**

I built Noetic from sources (desktop full) in Ubuntu jammy with minor changes:
1. Change in the CMakeLists.txt of some packages that use std::shred_mutex to use C++17 instead of C++11/C++14
2. Change `rosconsole` to use shared pointers instead of raw pointers

The ros1_bridge was compiled alone in a workspace, sourcing ROS1 workspace and ROS2 installation from binaries in `/opt/ros/iron`

- [X] **Run the bridge with talker and listener**

The messages were bridged. Only warning in the bridge related to `rosout`.

[image]

- [X] **Run the bridge and exchange images**

The messages were bridged.

[image]

- [X] **Run the bridge and AddTwoInts service**

The service successfully worked.

[image]

- [X] **Run the bridge with only selected topics and services.**

Both message publication and service were  working

[image]

@Yadunund @clalancette I think we can close this issue :nerd_face:

fmrico, 2023-05-15:
Hi @Yadunund 

I'm sorry to tell you that using the ROS1 binaries in Jammy, and compiling as described [here](https://docs.ros.org/en/humble/How-To-Guides/Using-ros1_bridge-Jammy-upstream.html#), I have not been able to get the bridge working. The `chatter` topic created by  `rosrun rospy_tutorials talker` does not appear in ROS 2. Maybe because:

```
$ ros2 run ros1_bridge dynamic_bridge --print-pairs
No message type conversion pairs supported.
No service type conversion pairs supported.
```

And message headers are installed:

```
$ dpkg -l | grep std-msgs
ii  libstd-msgs-dev:amd64                                    0.5.13-2                                amd64        C/C++ headers for Standard Robot OS Messages
ii  python3-std-msgs                                         0.5.13-2                                all          Python 3 interface for Standard Robot OS Messages
ii  ros-humble-std-msgs                                      4.2.3-1jammy.20230426.013042            amd64        A package containing some standard message definitions.
ii  ros-iron-std-msgs                                        5.0.0-2jammy.20230429.012734            amd64        A package containing some standard message definitions.
ii  ros-rolling-std-msgs                                     5.1.0-1jammy.20230512.165632            amd64        A package containing some standard message definitions.
ii  ros-std-msgs 
```

Best

Yadunund, 2023-05-16:
Thanks a lot for giving that a shot @fmrico! I've logged the issue [here](https://github.com/osrf/ros2_test_cases/discussions/1055) and will look into it with the team.

closed 2023-05-16 by Yadunund

Yadunund, 2023-05-19:
@fmrico in case you still have your worksapce intact, do you mind switching to cyclonedds (rmb `ros2 daemon stop`) to see if the chatter topic can be discovered? I'll update https://github.com/osrf/ros2_test_cases/issues/791 accordingly