osrf/ros2_test_cases #797: ros2cli network tests

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- DDS vendor: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fujitatomoya, 2023-05-09:
/assign @fujitatomoya

fujitatomoya, 2023-05-09:
### Verification Summary

No problem observed.

### Procedure

- Machine-A

```bash
export RMW_IMPLEMENTATION=rmw_connextdds
ros2 launch demo_nodes_cpp talker_listener_launch.py
```

- Machine-B

```bash
export RMW_IMPLEMENTATION=rmw_connextdds
ros2 node list
ros2 topic list
ros2 service list
ros2 param list
```

### Check Note

> Node discovery

✔️ 

> Topic discovery

✔️ 

> Service discovery

✔️ 

> Parameter discovery

✔️

fujitatomoya, 2023-05-09:
@Yadunund i think we can close this issue.

Yadunund, 2023-05-10:
Awesome, thanks for testing!