osrf/ros2_test_cases #800: ros2cli network tests

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-07:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Ubuntu Jammy

[image]

# Results

[image]

**ros2cli can communicate with nodes running on another machine. The check is passed.**

clalancette, 2023-05-07:
Thank you for testing!

closed 2023-05-07 by clalancette