osrf/ros2_test_cases #801: ros2cli network tests

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-07:
Hello could you please assign this issue to me? Thank you so much,

juliamp22, 2023-05-07:
**Node discovery** Works fine.

[image]

juliamp22, 2023-05-07:
**Topic discovery** - Works fine
[image]

juliamp22, 2023-05-07:
**Service discovery** - Works fine
[image]

juliamp22, 2023-05-07:
**Parameter discovery** - Works fine

[image]

juliamp22, 2023-05-07:
This Issue can be closed as all works. Thank you so much.

pradyum, 2023-05-07:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

### Machine 1 ( Running - talker_listener_launch.py )

[image]

### Machine 2 ( Testing checks )

[image]

# Result

[image]

**ros2cli in machine 2 is able to communicate with nodes running in machine 1. And all the checks are passed.**

**In Iron bera binary the distribution is showing as rolling.**

Hi @juliamp22, I appreciate your tests. Actually this test has be done on two different machine connected through networking.

juliamp22, 2023-05-07:
Perfect! @pradyum thank you so much for your recommendations.

juliamp22, 2023-05-07:
This Issue can be closed @clalancette . Thank you so much!

clalancette, 2023-05-08:
Thank you for testing!

closed 2023-05-08 by clalancette