osrf/ros2_test_cases #807: ros2cli network tests

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-04:
Tested with one laptop on Jammy binary, and a desktop on Windows 11 binary.
Tested both ways, first with turtlesim on Jammy and discovery commands on Windows, then with Turtlesim on Windows and discovery commands on Jammy.

No issue found on both cases, nodes/topics/service/parameter discovery cli commands worked as expected through the local wifi network.

clalancette, 2023-05-04:
Thanks for testing!

closed 2023-05-04 by clalancette

bai-ivan, 2023-05-06:
Just adding that testing over stationary win10 binary <>laptop over wifi windows 11 communication also works quite fine :)