osrf/ros2_test_cases #811: ros2cli network tests

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-10:
On 2 debian jammy installation with fastDDS.

Running the demo_node_cpp talker_listener_launch.py on another jammy laptop on the same subnet.
All ros2 commands on the main laptop properly discovers the nodes/topics/services/params of the other laptop.
After stopping the other laptop launch commands, the main laptop properly no more discover anything.

All Ok.

[image]

Yadunund, 2023-05-10:
> On 2 debian jammy installation with fastDDS.

Sorry I reopened because the Setup says `rhel-9` not `jammy`. Can I confirm your setup?

BorisBoutillier, 2023-05-10:
Oops, my bad, I mis-read this testcase. This  was not on a RHEL-9, sorry.

This testcase is still open for testing, I don't have access to a RHEL-9

Yadunund, 2023-05-10:
No problem! Thanks for confirming

pradyum, 2023-05-15:
# Setup

- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

[image]

[image]

# Result

**The nodes are discoverable and able to get expected output for the checks. All checks are passed.**

clalancette, 2023-05-15:
Thanks for testing!

closed 2023-05-15 by clalancette