osrf/ros2_test_cases #812: ros2cli network tests

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-07:
Working on this (it's second active , but i'm on 40th minute of compiling the bridge, so i'd do another while i wait)

bai-ivan, 2023-05-07:
1) works
```
C:\Users\User>ros2 node list
/minimal_publisher

C:\Users\User>ros2 node list

```
2) works
```
C:\Users\User>ros2 topic list
/parameter_events
/rosout
/topic

C:\Users\User>ros2 topic list
/parameter_events
/rosout
```
3) works
```
C:\Users\User>ros2 topic list
/parameter_events
/rosout
/topic

C:\Users\User>ros2 topic list
/parameter_events
/rosout
```
4) works
```
C:\Users\User>ros2 param list
/introspection_client:
  client_configure_introspection
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time
/introspection_service:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  service_configure_introspection
  use_sim_time

C:\Users\User>ros2 param list
```

Yadunund, 2023-05-07:
Thanks for testing!

closed 2023-05-07 by Yadunund