osrf/ros2_test_cases #820: ros2cli on a local machine

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Help text**

     Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 --help
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic --help
       ```
  
    **Expect**
    
    1.
       Long-form help is available.
    
  </details>
- [ ] **Tab completion**

     When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Lists the commands
       ```bash
       # User input in terminal 1
       ros2 <tab><tab>
       ```
    1.
       Completes the word `launch`
       ```bash
       # User input in terminal 1
       ros2 lau<tab>
       ```
    1.
       Completes until `demo_nodes_`
       ```bash
       # User input in terminal 1
       ros2 launch dem<tab>
       ```
    1.
       Completes the package `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_c<tab>
       ```
    1.
       Lists launch files in `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp <tab><tab>
       ```
    1.
       Suggests the launch files name `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp ta<tab>
       ```
    1.
       Lists several launch files that start with `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener<tab><tab>
       ```
    1.
       Completes the launch files name `talker_listener.launch.py`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.p<tab>
       ```
  
    **Expect**
    
    1.
       You eventually get to the full command
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
  </details>
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 node list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /listener
       /talker
       ```
  </details>
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /chatter
       /parameter_events
       /rosout
       ```
  </details>
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /add_two_ints
       /add_two_ints_server/describe_parameters
       /add_two_ints_server/get_parameter_types
       /add_two_ints_server/get_parameters
       /add_two_ints_server/list_parameters
       /add_two_ints_server/set_parameters
       /add_two_ints_server/set_parameters_atomically
       ```
  </details>
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /turtlesim:
         background_b
         background_g
         background_r
         qos_overrides./parameter_events.publisher.depth
         qos_overrides./parameter_events.publisher.durability
         qos_overrides./parameter_events.publisher.history
         qos_overrides./parameter_events.publisher.reliability
         use_sim_time
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-07:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

[image]

# Result

[image]

**Everything works except the tab completion.**

clalancette, 2023-05-07:
We have some other open issues about the tab completion, so I'm going to mark this as complete.

Thank you for testing!

closed 2023-05-07 by clalancette