osrf/ros2_test_cases #828: ros2cli on a local machine

Yadunund, 2023-04-05:
Test the basic functionality of the ros2cli package family

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Help text**

     Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 --help
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic --help
       ```
  
    **Expect**
    
    1.
       Long-form help is available.
    
  </details>
- [ ] **Tab completion**

     When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Lists the commands
       ```bash
       # User input in terminal 1
       ros2 <tab><tab>
       ```
    1.
       Completes the word `launch`
       ```bash
       # User input in terminal 1
       ros2 lau<tab>
       ```
    1.
       Completes until `demo_nodes_`
       ```bash
       # User input in terminal 1
       ros2 launch dem<tab>
       ```
    1.
       Completes the package `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_c<tab>
       ```
    1.
       Lists launch files in `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp <tab><tab>
       ```
    1.
       Suggests the launch files name `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp ta<tab>
       ```
    1.
       Lists several launch files that start with `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener<tab><tab>
       ```
    1.
       Completes the launch files name `talker_listener.launch.py`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.p<tab>
       ```
  
    **Expect**
    
    1.
       You eventually get to the full command
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
  </details>
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 node list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /listener
       /talker
       ```
  </details>
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /chatter
       /parameter_events
       /rosout
       ```
  </details>
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /add_two_ints
       /add_two_ints_server/describe_parameters
       /add_two_ints_server/get_parameter_types
       /add_two_ints_server/get_parameters
       /add_two_ints_server/list_parameters
       /add_two_ints_server/set_parameters
       /add_two_ints_server/set_parameters_atomically
       ```
  </details>
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /turtlesim:
         background_b
         background_g
         background_r
         qos_overrides./parameter_events.publisher.depth
         qos_overrides./parameter_events.publisher.durability
         qos_overrides./parameter_events.publisher.history
         qos_overrides./parameter_events.publisher.reliability
         use_sim_time
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-05:
Hey, I would like to work on this ticket

Kartik9250, 2023-05-05:
## Checks
- [x] Help text
- [x] Tab completion
- [x] Node discovery
- [x] Topic discovery
- [x] Service discovery
- [x] Parameter discovery

<br>

Everything worked flawlessly, just wanted to point out two things:
- The `talker_listener.launch.py` should be changed to `talker_listener_launch.py`
>   8. Completes the launch files name `talker_listener.launch.py`
>      ```shell
>      # User input in terminal 1
>      ros2 launch demo_nodes_cpp talker_listener.launch.p<tab>
>      ```

- The output of `ros2 param list` after running the `ros2 run turtlesim turtlesim_node` also gave an `holonomic` parameter
>   1. ```shell
>      # stdout in terminal 2
>      /turtlesim:
>        background_b
>        background_g
>        background_r
>        qos_overrides./parameter_events.publisher.depth
>        qos_overrides./parameter_events.publisher.durability
>        qos_overrides./parameter_events.publisher.history
>        qos_overrides./parameter_events.publisher.reliability
>        use_sim_time
>      ```
>

clalancette, 2023-05-05:
> * The `talker_listener.launch.py` should be changed to `talker_listener_launch.py`

Thanks.  We've already fixed that one for next year in https://github.com/audrow/yatm/pull/347.

> * The output of `ros2 param list` after running the `ros2 run turtlesim turtlesim_node` also gave an `holonomic` parameter

Good call.  I've opened a PR to fix this in https://github.com/audrow/yatm/pull/350

Since all of the issues are cosmetic ones, I'm going to go ahead and close this.  Thanks for testing!

closed 2023-05-05 by clalancette