osrf/ros2_test_cases #836: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-09:
Under details of this issue: `ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py` ->  `ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py`.

I was able to launch the dummy_robot_bringup, but unable to see the sine wave in rqt.
[image]

After shutting down rqt, I got this ClassLoader error:
[image]

Yadunund, 2023-05-10:
Thanks for testing! 

* The launch file name was an error and has been fixed for future test cases that will be generated
* the rqt_plot bug has been fixed https://github.com/ros-visualization/rqt_plot/pull/90
* the rcl_shutdown exception is already being looked at. 
I'll close this ticket as we're following up on any outstanding issues else where

closed 2023-05-10 by Yadunund