osrf/ros2_test_cases #838: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Connext
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-14:
Happens the same as in [#861 ](https://github.com/osrf/ros2_test_cases/issues/861)
The node can be executed in terminal 1 with success and rqt can be opened with success but, the problem is that in the terminal 2 no feedback can be obtained from terminal 1. So, in rqt neither feedback appears.

Yadunund, 2023-05-15:
Thanks for trying this out.
To clarify, when you say no feedback, are you referring to rqt_plot not visualizing the joint state? If so, could you try with the changes introduced here 
https://github.com/ros-visualization/rqt_plot/pull/90

juliamp22, 2023-05-15:
I was referring that when I made the ros2 topic list / ros2 node list in another terminal I am not able to see any feedback from the nodes launched in terminal1.

Yadunund, 2023-05-16:
I see. Does it work with a different RMW implmentation?

juliamp22, 2023-05-16:
Hello @Yadunund yes, I tried with **rmw_cyclonedds_cpp** it gives me the following error: 
`RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_iron\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found`

[image]

`In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to /joint_states/position[0]. You should see a Sine wave.` Does not work 

`With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.`  It appears a dummy_joint_states that publishes the joint states to the robot_state_publisher

With 'Plugins > Logging > Bag' you should be able record a ROS Bag file. It let me record a ros Bag file. 

With 'Plugins > Topic > Topic Monitor' you should be able to check the /joint_states topic and view messages streaming in. Appears this error: Error calling timeline_changed on <class 'rqt_bag.player.Player'>: 'Player' object has no attribute 'timeline_changed'

[image]

juliamp22, 2023-05-16:
Testing again with connextdds: 

_In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to /joint_states/position[0]. You should see a Sine wave._ Does not work, it does not let me add any topic.

`With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot`. It appearsnothing

`With 'Plugins > Logging > Bag' you should be able record a ROS Bag file`. It let me record a ros Bag file.

With 'Plugins > Topic > Topic Monitor' you should be able to check the /joint_states topic and view messages streaming in.  It does not appears anyting. 

[image]

Yadunund, 2023-05-18:
We fixed the rqt_plot issue recently so it would be good to pull in the changes from https://github.com/ros-visualization/rqt_plot/pull/91

juliamp22, 2023-05-18:
Same source installation errors as [861](https://github.com/osrf/ros2_test_cases/issues/861)

clalancette, 2023-06-23:
With the tutorial party over, I'm going to close this out.  Thanks for your efforts here.

closed 2023-06-23 by clalancette