osrf/ros2_test_cases #840: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-07:
Hello @clalancette @Yadunund could you please assign this task to me? Thank you so much!

juliamp22, 2023-05-07:
Works fine: 
[image]

juliamp22, 2023-05-07:
@clalancette @Yadunund could you please close the issue? Thank you so much

pradyum, 2023-05-07:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

[image]

# Results

[image]

**The same issue and result as seen in https://github.com/osrf/ros2_test_cases/issues/847**

**RQT plot bug is fix in https://github.com/ros-visualization/rqt_plot/pull/90**

**RQT bag issue is added here https://github.com/osrf/ros2_test_cases/discussions/1048**

Hi @juliamp22 the above issue have be there in beta release. The RQT plot bug is fixed and RQT bag bug is added to discussion.

juliamp22, 2023-05-07:
Lots of thanks @pradyum!

Yadunund, 2023-05-08:
Thanks @juliamp22 and @pradyum for testing! I'll close this since issues have been documented. I've updated the Discussion to include the warnings.

closed 2023-05-08 by Yadunund