osrf/ros2_test_cases #842: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Abishalini, 2023-05-03:
I'll test this

Abishalini, 2023-05-03:
[image]

I was unable to visualize the sine plot when I set the topic to `/joint_states/position[0]`
But I was able to visualize the node graph, record a bag and could monitor topics.

Abishalini, 2023-05-03:
I was not able to shut `rqt` using `Ctrl + C`. Looks like its stuck in a deadlock because of
```
^C[INFO] [1683141198.559929094] [rclcpp]: signal_handler(signum=2)
rclpy.shutdown() was called before QThread.quit()
Error recording to bag: failed to get topic names and types: rcl node's context is invalid, at ./src/rcl/node.c:434
```

Abishalini, 2023-05-03:
Ran `rqt` again and found the following error on shutdown
```
abishalini@abishalini:~/ws_iron_testing$ rqt
^C[INFO] [1683141802.053946428] [rclcpp]: signal_handler(signum=2)
rclpy.shutdown() was called before QThread.quit()
PluginHandler.shutdown_plugin() plugin "rqt_plot/Plot#1" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 136, in shutdown_plugin
    self._shutdown_plugin()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler_direct.py", line 92, in _shutdown_plugin
    self.emit_shutdown_plugin_completed()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 150, in emit_shutdown_plugin_completed
    callback(self._instance_id)
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 480, in _close_application_shutdown_callback
    self._close_application_signal()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 483, in _close_application_signal
    self._plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 62, in shutdown
    self._destroy_node()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 89, in _destroy_node
    rclpy.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown
    _shutdown(context=context)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:313

^C[INFO] [1683141846.725957543] [rclcpp]: signal_handler(signum=2)
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 127 in ./src/class_loader.cpp

```

Yadunund, 2023-05-04:
> I was unable to visualize the sine plot when I set the topic to /joint_states/position[0]

Thanks a lot for identifying this bug. It looks like a regression so I've opened a ticket https://github.com/ros-visualization/rqt_plot/issues/89

Yadunund, 2023-05-18:
Closing this one out since we're patched all the RQT bugs except the rqt_bag one which has been ticketed.