osrf/ros2_test_cases #847: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Chip: Arm64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-05:
1. In 'Plugins > Visualization > Plot' plugin, I am unable to add the topic since the PLUS button is always disabled as shown in the image
[image]

2. In 'Plugins > Logging > Bag' plugin, I am able to record the topics and save to .db3 file. But, I am not able to open the .db3 file which shows as unsupported file format to open as shown in the image
[image]

Other-than these problems the RQT runs as expected.

Yadunund, 2023-05-05:
Thanks for testing!

> In 'Plugins > Visualization > Plot' plugin, I am unable to add the topic since the PLUS button is always disabled as shown in the image

This is a bug that has been fixed yesterday https://github.com/ros-visualization/rqt_plot/pull/90. if you try the latest source version of `rqt_plot`, the test should work.

> In 'Plugins > Logging > Bag' plugin, I am able to record the topics and save to .db3 file. But, I am not able to open the .db3 file which shows as unsupported file format to open as shown in the image

This sounds like a bug. Firstly i would expect the bag to be an mcap file and not db3. Second, the bag folder is missing the `metadata.yaml` file

Yadunund, 2023-05-05:
Looks like the metadata bug has been documented https://github.com/ros-visualization/rqt_bag/issues/127. Without that file, it's not possible to open a bag file.

I'll close this ticket as other rqt features work. I've added the rqt_bag issue here https://github.com/osrf/ros2_test_cases/discussions/1048 so that we can keep track of it.

closed 2023-05-05 by Yadunund