osrf/ros2_test_cases #848: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-05:
Hey, I would like to work on this ticket

Kartik9250, 2023-05-05:
[image]

- I was unable to visualise the sine wave using Plot.
- Node Graph worked properly.
- Ros bag works but the data(the green lines) updating in gui felt laggy. I viewed the data it seemed okay.
- Topic Monitor is working.

Yadunund, 2023-05-05:
> I was unable to visualise the sine wave using Plot.

This is a bug that has been fixed yesterday https://github.com/ros-visualization/rqt_plot/pull/90. if you try the latest source version of rqt_plot, the test should work.

> I viewed the data it seemed okay.

Could I check if you were able to save the bag you recorded, and load it without any errors? An error regarding opening the bag was reported here https://github.com/osrf/ros2_test_cases/issues/847#issuecomment-1535635022

Kartik9250, 2023-05-05:
@Yadunund No i am unable to load the file in rosbag in rqt. It gives out these error: 
`[ERROR] [1683310232.381725064] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI
Loading 'b'/home/brad/2023-05-05-20-01-23'' failed due to: No storage could be initialized from the inputs.`

Infact it does not even create a `.yaml` file like the rosbag cli does. The cli version also used `.mcap` while rqt saves bag in `.db3`

And sorry, I must have forgotten to mention this last time that there is no `stop button` as such while recording the bag file, the `save button` remains disabled while recording. I just closed the rqt last time and thought that did it. These warnings were also prompted on the terminal:
`[WARN] [1683310261.285516513] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1683310261.355283277] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None`

Yadunund, 2023-05-05:
Thanks for confirming. I'll close this ticket given that we're tracking the rqt_bag bug elsewehre.

closed 2023-05-05 by Yadunund