osrf/ros2_test_cases #849: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Chip: Arm64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Barry-Xu-2018, 2023-05-06:
My test environment
- Hardware 
    - Raspberry Pi 4 Model B
- Software
    - Ubuntu 22.04.2 LTS
        ```bash
        ubuntu@ubuntu:~$ uname  -a
        Linux ubuntu 5.15.0-1027-raspi #29-Ubuntu SMP PREEMPT Mon Apr 3 10:12:21 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
        
        ubuntu@ubuntu:~$ cat /etc/lsb-release 
        DISTRIB_ID=Ubuntu
        DISTRIB_RELEASE=22.04
        DISTRIB_CODENAME=jammy
        DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
        ```

Following above test steps, I can run commands successfully without any error.

Test items and result

- In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to /joint_states/position[0]. You should see a Sine wave. `Fail` :x:

    Nothing is showed.  
    [image]

- With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.  `Pass` :heavy_check_mark:
 
    [image]
   
- With 'Plugins > Logging > Bag' you should be able record a ROS Bag file. `Pass` :heavy_check_mark:
 
    [image]

- With 'Plugins > Topic > Topic Monitor' you should be able to check the /joint_states topic and view messages streaming in. `Pass` :heavy_check_mark:

    [image]

pradyum, 2023-05-06:
Hi, I did the test in debian installation. Still the issue is there as said in https://github.com/osrf/ros2_test_cases/issues/847.

Yadunund, 2023-05-08:
Thanks for testing! Closing as the rqt_plot issue has been documented and fixed in source.

closed 2023-05-08 by Yadunund