osrf/ros2_test_cases #850: RQT

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-06:
**Setup**

- DDS vendor: FastDDS
- BuildType: Source
- Chip: Amd64
- OS: Rhel 9

Plot plugin had a issue

[image]

The other issue explained in https://github.com/osrf/ros2_test_cases/issues/847 is observed.

Other checks are passed.

Yadunund, 2023-05-08:
cc @cottsay

cottsay, 2023-05-09:
We don't currently have those packages in RHEL 9, but we will soon. `python3-matplotlib` was submitted for testing just now, and should go stable in about a week: https://bodhi.fedoraproject.org/updates/FEDORA-EPEL-2023-76a7a89a90

I'll need to update the release repository to un-skip the dependency, but that should unblock this scenario.

Yadunund, 2023-05-19:
Thanks for the heads up @cottsay! I'll close this ticket.

closed 2023-05-19 by Yadunund