osrf/ros2_test_cases #857: Rviz

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-13:
Hello @Yadunund I can open rviz2 without any problem but, there is a problem getting feedback from launch node dummy_robot_bringup_launch.py. When I tried ros2 topic list it gives nothing:

[image]

Yadunund, 2023-05-15:
Thanks for trying this out.

Can I clarify if you're testing with windows FAT binaries downloaded from the archives? I would not have expect any node to run given the issue reported here https://github.com/osrf/ros2_test_cases/issues/711

It seems for now, we can only support connectdds if ROS 2 is built from source on windows

juliamp22, 2023-05-15:
Hello @Yadunund I was testing with windows binaries. I am trying to install ros2 to windows from source, but I get this error:

CMake Error at C:/dev/ros2_source/build/orocos_kdl_vendor/orocos_kdl-507de66-prefix/src/orocos_kdl-507de66-build/orocos_kdl-config.cmake:13 (include):
  include could not find requested file:
C:/dev/ros2_source/build/orocos_kdl_vendor/orocos_kdl-507de66-prefix/src/orocos_kdl-507de66-build/FindEigen3.cmake

Have you seen this error before? Thank you so much.

[image]

Yadunund, 2023-05-16:
Any chance you don't have Eigen3 on your machine? It should get installed as part of this [step]( https://docs.ros.org/en/iron/Installation/Alternatives/Windows-Development-Setup.html#install-dependencies)

juliamp22, 2023-05-16:
Hello @Yadunund I found that the solution is uninstalling and reinstalling eigen.
Now I get the following error: 

 1 package failed: mimick_vendor

[image]

juliamp22, 2023-05-16:
After multiple reattempts, (closing and opening cmds) it worked: 
[image]

Yadunund, 2023-05-18:
Thanks for testing! I'm assuming you resolved the mimic_vendor failed build.
I've noticed that sometimes with windows I need to manually hit "Enter" in the terminal a few extra times for the process to start running 🤷🏼 
I'll close this ticket for now

closed 2023-05-18 by Yadunund

juliamp22, 2023-05-18:
Hello @Yadunund finally I did not need to install the windows source to run the rviz with _rmw_cyclonedds_cpp_. I just try again to run the steps with a little more love and it worked! Yes, sometimes in windows you have be patient (leave enough time to open the packages) and also hit manually "Enter" in the terminal a few extra times for the process to start running.

Yadunund, 2023-05-19:
Thanks for the patience with testing!