osrf/ros2_test_cases #861: Rviz

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Connext
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-14:
The node can be executed in terminal 1 with success and rviz can be opened with success but, the problem is that in the terminal 2 no feedback can be obtained from terminal 1. So, in rviz2 neither feedback appears.

[image]

[image]

juliamp22, 2023-05-18:
Cannot install Windows source packages due to a mimick error:

[image]

If I clone directly the mimick repository and try to build it, it generates the following error:

[image]

Yadunund, 2023-05-19:
Hmm that is strange as other users have been able to build windows from source although it takes more time.

Just to check, you have the latest cmake installed via choco by following [this step](https://docs.ros.org/en/iron/Installation/Alternatives/Windows-Development-Setup.html#install-dependencies)? And that you unchecked installing any cmake tooks when installing visual studio in [this step](https://docs.ros.org/en/iron/Installation/Alternatives/Windows-Development-Setup.html#install-visual-studio)?

Could you try again with a clean build? (delete the build and install directories). Also, can I check if you were building with `colcon build --merge-install` ?

juliamp22, 2023-05-19:
I uninstall windows 11 and reinstall windows 10. I have seen that this gives lots of problems in windows 11, maybe that could explain why.  All your suggestions are fulfilled.
I've tried so hard to install all, without success. I do not know if someone else of the community can help. Sorry for the inconvenience.

clalancette, 2023-06-23:
With the tutorial party over, I'm going to close this out.  Thanks for your efforts here.

closed 2023-06-23 by clalancette