osrf/ros2_test_cases #863: Rviz

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-07:
Hello @clalancette @Yadunund could you please assign this issue to me? Thank you so much.

juliamp22, 2023-05-07:
Appears this _System Error_: 

[image]

pradyum, 2023-05-07:
# Setup

- DDS vendor: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

[image]

# Result

[image]

**When I tried to run rviz2 i got this error "zlib1.dll was not found". And this is missing in rviz_ogre_vendor/bin folder**

[image]

**After I placed the compiled zlib1.dll file in rviz_orge_vendor/bin folder. Then rviz works as expected.**

Also the laserscan status flickering issue happen on first message when toggled and stop flickering.

juliamp22, 2023-05-07:
Hello @pradyum you are totally right I copied the _zlib1.dll_ located in ros2_iron\Lib\rviz2 to _ros2_iron\opt\rviz_ogre_vendor\bin_ and now works perfectly! 

[image]

juliamp22, 2023-05-07:
This issue can be closed @clalancette .  Thank you so much!

Yadunund, 2023-05-08:
> After I placed the compiled zlib1.dll file in rviz_orge_vendor/bin folder. Then rviz works as expected.

Thanks for identifying the problem but this would not be the right fix.

The Windows binaries installation page is missing these instructions to add zlib binary to the system path. That way you don't need to copy the .dll into the binary folder of each package that links against it. I'll open a PR to the documentation.

juliamp22, 2023-05-08:
Hello @Yadunund thank you very much for your indications.
After downloading and installing zlib following `[this ](https://docs.ros.org/en/iron/Installation/Alternatives/Windows-Development-Setup.html#install-miscellaneous-prerequisites) `instructions and removing the zlib1.dll previously located in ros2_iron\Lib\rviz2 to ros2_iron\opt\rviz_ogre_vendor\bin.

I am available to run rviz2:

[image]

The Issue can be closed! Thank you so much.

Yadunund, 2023-05-09:
Great, thanks for testing!

closed 2023-05-09 by Yadunund