osrf/ros2_test_cases #865: Rviz

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Abishalini, 2023-05-03:
I'm testing this now

Abishalini, 2023-05-03:
I was able to load the robot and visualize the laser scan 
https://user-images.githubusercontent.com/17363793/236015574-07225eb7-8f62-486f-9303-799f9d8dd9b2.mp4

Additionally, I tested MoveIt's Motion Planning panel and I don't see any issues
https://user-images.githubusercontent.com/17363793/236015956-67838eaf-dcec-4578-b4e6-b143b6c0207d.mp4

clalancette, 2023-05-03:
Thank you for testing!

closed 2023-05-03 by clalancette