osrf/ros2_test_cases #868: Rviz

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Chip: Amd64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

iuhilnehc-ynos, 2023-05-06:
Use
`ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py`
instead of
`ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py`.

<details><summary> log </summary>
<p>

# terminal 1

```shell
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/chenlh/.ros/log/2023-05-05-16-13-43-514918-OptiPlex-7080-1641639
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [1641653]
[INFO] [robot_state_publisher-2]: process started with pid [1641655]
[INFO] [dummy_joint_states-3]: process started with pid [1641657]
[INFO] [dummy_laser-4]: process started with pid [1641659]
[dummy_laser-4] [INFO] [1683274423.710153764] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1683274423.710216758] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1683274423.710224097] [dummy_laser]: scan time increment: 	0.000028
[robot_state_publisher-2] [INFO] [1683274423.714457346] [robot_state_publisher]: got segment single_rrbot_camera_link
[robot_state_publisher-2] [INFO] [1683274423.714513797] [robot_state_publisher]: got segment single_rrbot_hokuyo_link
[robot_state_publisher-2] [INFO] [1683274423.714523662] [robot_state_publisher]: got segment single_rrbot_link1
[robot_state_publisher-2] [INFO] [1683274423.714529837] [robot_state_publisher]: got segment single_rrbot_link2
[robot_state_publisher-2] [INFO] [1683274423.714535203] [robot_state_publisher]: got segment single_rrbot_link3
[robot_state_publisher-2] [INFO] [1683274423.714539993] [robot_state_publisher]: got segment world
```

# terminal 2

```shell
$ rviz2
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[INFO] [1683274444.811177187] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1683274444.811222659] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1683274444.821606556] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1683274840.790165800] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1683274840.806912 but the latest data is at time 1683274840.789674, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [1683274840.874816336] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1683274840.873580 but the latest data is at time 1683274840.849708, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
[ERROR] [1683274840.971003047] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1683274840.973723 but the latest data is at time 1683274840.969704, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
...
```

</p>
</details>

# Expect

- You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
OK
- You should see a laser scan emitting from the waving end of the dummy robot.
OK
- Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
NG

# Problems
After unchecking and checking the `LaserScan` checkbox, the `LaserScan` status in the Displays becomes unstable (switch state repeatedly Error <-> Ok ).
rviz2 in the console outputs error information frequently.

[image]
[image]
[image]

# Fixed?

https://github.com/ros2/rviz/issues/852#issuecomment-1523704615

hwoithe, 2023-05-07:
For what it's worth, I am also seeing the same behavior.

Yadunund, 2023-05-08:
Thanks for highlighting the issue. I've opened this discussion to keep track of it https://github.com/osrf/ros2_test_cases/discussions/1053

Since other rviz functionality works, I'll close this ticket.

closed 2023-05-08 by Yadunund