osrf/ros2_test_cases #869: Rviz

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-04:
Done on a Windows 11 binary installation.

- Rviz2 opens correctly
- The dummy robot's kinematic tree is visible
- I can add and see the LaserScan topic
-  I have played around with camera, perspective, and everything is smooth and stable.

**_Note_**:  The test_case launch file name in the command is incorrect, should be ` dummy_robot_bringup_launch.py`  and not `dummy_robot_bringup.launch.py`

Screenshot in action:
[image]

clalancette, 2023-05-04:
> **_Note_**: The test_case launch file name in the command is incorrect, should be ` dummy_robot_bringup_launch.py` and not `dummy_robot_bringup.launch.py`

Thanks for the note.  We've fixed that already for next year, but we didn't want to go and regenerate all of the test cases.  So you'll likely see this elsewhere as well.

In any case, thanks for the testing!  Closing this out.

closed 2023-05-04 by clalancette