osrf/ros2_test_cases #872: Rviz

Yadunund, 2023-04-05:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Chip: Arm64
- Os: Ubuntu Jammy

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jjd9, 2023-05-05:
> ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py

This is not correct. `dummy_robot_bringup.launch.py` does not exist after I installed ROS2. The launch file I have on my system is called `dummy_robot_bringup_launch.py`.

After messing around with rviz2 for a while, I did not see any issues with the visualization for this test case.

Default
[image]

Having fun with the /scan topic settings (-:
[image]

However, if i change the QoS setings of the laser scan (e.g. changing the reliability from reliable to best effort, it causes TF2 to spam this error). Yet the /scan visualization is fine (aside from the topic field flickering in the Display panel).
```
[ERROR] [1683258196.510416507] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1683258196.532693 but the latest data is at time 1683258196.509739, when looking up transform from frame [single_rrbot_hokuyo_link] to frame [world]
```
Even if i switch the QoS back to the default settings, it continues to spam the error. But if I remove the /scan topic and re-add it, that fixes the error. so maybe an issue with the bookkeeping for the topic?

Kartik9250, 2023-05-05:
Hey, I would like to take on this ticket

Barry-Xu-2018, 2023-05-06:
My test environment
- Hardware 
    - Raspberry Pi 4 Model B
- Software
    - Ubuntu 22.04.2 LTS

Following above test steps, I can run commands successfully without any error.

Test items and result

-  You should see the dummy robot's kinematic tree waving back and forth in the rviz window.   `PASS`

- You should see a laser scan emitting from the waving end of the dummy robot. `PASS`

- Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable. `PASS`

@Kartik9250 You can continue to do test on your environment. Maybe you can find unstable cases.