osrf/ros2_test_cases #875: Sane development environment

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Test message generation with .msg, .srv, .action, and .idl on top of the installed environment**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-07:
Hello @clalancette @Yadunund , does it refers to this tutorials?
-  https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html 
- https://docs.ros.org/en/iron/Tutorials/Intermediate/Creating-an-Action.html
Thank you so much

clalancette, 2023-05-07:
> * https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html
> 
> * https://docs.ros.org/en/iron/Tutorials/Intermediate/Creating-an-Action.html

Yes, if you go through those tutorials, that will cover this test.

juliamp22, 2023-05-08:
Following [this](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html) tutorial. I could generate with success the following:

- msg and srv:

[image]

Testing the generated msg as a dependency of a node publisher and subscriber:

[image]

Testing the srv as a dependency of a node server and client:

[image]

juliamp22, 2023-05-08:
- Actions also works fine: 

[image]

juliamp22, 2023-05-08:
This Issue can be close as it works fine! Thank you so much.

clalancette, 2023-05-09:
Thank you for testing!

closed 2023-05-09 by clalancette