osrf/ros2_test_cases #876: Sane development environment

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Test message generation with .msg, .srv, .action, and .idl on top of the installed environment**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Robotgir, 2023-05-11:
please assign

Robotgir, 2023-05-15:
1. installed cyclone dds with
"sudo apt install ros-iron-rmw-cyclonedds-cpp"
2. changed to cyclone dds with command "export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp"
3. checked if right dds in use with command "echo $RMW_IMPLEMENTATION"

can confirm that the c cpp py files are generated for all message types like .msg, .srv, .action and .idl and the examples used to communicate via these message types work as expected.