osrf/ros2_test_cases #877: Sane development environment

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Test message generation with .msg, .srv, .action, and .idl on top of the installed environment**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
working on this.

bai-ivan, 2023-05-06:
a note while I'm at it : tried including std_msgs's header so it at least looks a bit like a sane message but failed with some weird error. I honestly haven't build custom msgs under windows, i usually use wsl2 + whatever ubuntu is needed inside. Posting it here in case it's useful for someone.

```
[12.422000] (test_msgs) StdoutLine: {'line': b'  Generating type hashes for ROS interfaces\r\n'}
[12.469000] (-) TimerEvent: {}
[12.579000] (-) TimerEvent: {}
[12.688000] (-) TimerEvent: {}
[12.719000] (test_msgs) StdoutLine: {'line': b'  Traceback (most recent call last):\r\n'}
[12.719000] (test_msgs) StdoutLine: {'line': b'    File "C:/dev/ros2-iron/lib/rosidl_generator_type_description/rosidl_generator_type_description", line 50, in <module>\r\n'}
[12.719000] (test_msgs) StdoutLine: {'line': b'      sys.exit(main())\r\n'}
[12.719000] (test_msgs) StdoutLine: {'line': b'    File "C:/dev/ros2-iron/lib/rosidl_generator_type_description/rosidl_generator_type_description", line 46, in main\r\n'}
[12.719000] (test_msgs) StdoutLine: {'line': b'      generate_type_hash(args.generator_arguments_file)\r\n'}
[12.719000] (test_msgs) StdoutLine: {'line': b'    File "C:\\dev\\ros2-iron\\Lib\\site-packages\\rosidl_generator_type_description\\__init__.py", line 160, in generate_type_hash\r\n'}
[12.719000] (test_msgs) StdoutLine: {'line': b'      pkg_dir = include_map[pkg]\r\n'}
[12.719000] (test_msgs) StdoutLine: {'line': b"  KeyError: 'std_msgs'\r\n"}
[12.750000] (test_msgs) StdoutLine: {'line': b"C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\MSBuild\\Microsoft\\VC\\v160\\Microsoft.CppCommon.targets(241,5): error MSB8066: Custom build for 'C:\\Users\\user\\test\\build\\test_msgs\\CMakeFiles\\c93cbfced6c4eb2211a0873498af8542\\Test.json.rule;C:\\Users\\user\\test\\build\\test_msgs\\CMakeFiles\\5af2a3684d4a6b4b164b36e9870bd7dc\\test_msgs__rosidl_generator_type_description.rule;C:\\Users\\user\\test\\src\\test_msgs\\CMakeLists.txt' exited with code 1. [C:\\Users\\user\\test\\build\\test_msgs\\test_msgs__rosidl_generator_type_description.vcxproj]\r\n"}
[12.766000] (test_msgs) CommandEnded: {'returncode': 1}
[12.797000] (-) TimerEvent: {}
```

gonna try without adding anything, just plain ints / bools to see if it compiles.

bai-ivan, 2023-05-06:
Works, generated them all. Not sure why it failed using std_msgs, but probably i'm no the proper person to debug this. 

msg file
```
#testmsg
int32 someint
bool somebool
float64 somefloat
```
result idl 
```
// generated from rosidl_adapter/resource/msg.idl.em
// with input from test_msgs/msg/Test.msg
// generated code does not contain a copyright notice

module test_msgs {
  module msg {
    @verbatim (language="comment", text=
      "testmsg")
    struct Test {
      int32 someint;

      boolean somebool;

      double somefloat;
    };
  };
};
```

srv file : 
```
int32 something
---
bool someresult
```

generated idl : 
```
// generated from rosidl_adapter/resource/srv.idl.em
// with input from test_msgs/srv/Test.srv
// generated code does not contain a copyright notice

module test_msgs {
  module srv {
    struct Test_Request {
      int32 something;
    };
    struct Test_Response {
      boolean someresult;
    };
  };
};
```
action msg:
```
int32 actiontest
---
int32 actionresult
---
int32 actionfeedback
```
Action idl : 
```
// generated from rosidl_adapter/resource/action.idl.em
// with input from test_msgs/action/Test.action
// generated code does not contain a copyright notice

module test_msgs {
  module action {
    struct Test_Goal {
      int32 actiontest;
    };
    struct Test_Result {
      int32 actionresult;
    };
    struct Test_Feedback {
      int32 actionfeedback;
    };
  };
};
```

Yadunund, 2023-05-08:
Thanks for testing this! Really helpful. Is a similar message thrown when you include another message type? eg `sensor_msgs/msg/LaserScan`?

Yadunund, 2023-05-08:
Thanks for testing this! Really helpful.
For a sanity check, did you include a `find_package(std_msgs REQUIRED)`and pass `std_msgs` as dependency to `rosidl_generate_interfaces`?

Eg : https://github.com/ros2/common_interfaces/blob/04c024e3d9d58ddb9b4843f903a12c9f2e8695c1/geometry_msgs/CMakeLists.txt#L16 and https://github.com/ros2/common_interfaces/blob/04c024e3d9d58ddb9b4843f903a12c9f2e8695c1/geometry_msgs/CMakeLists.txt#L51

bai-ivan, 2023-05-08:
@Yadunund I did, but the binary installation did not provide me with std_msgs. 
I am unsure on which of the other issue i tested compiling a node, and managed to compile the std_msgs from source, then compile a simple publisher node using rclcpp (single class publisher with timer, similar to the example one) where i depended on std_msgs and it worked flawlessly.

Yadunund, 2023-05-08:
> @Yadunund I did, but the binary installation did not provide me with std_msgs.

What is the output of `ros2 interface show std_msgs/msg/Header` ?

Also, I just observed that the name of your interface package is `test_msgs`. Such a package should already exist in your underlay fromhttps://github.com/ros2/rcl_interfaces/tree/rolling/test_msgs which may be causing the build issue. Maybe you can try again after renaming to something else.

I just tried building and running these two pkgs on Windows and was able to run the node without any issues. The msgs pkg links to std_msgs and geometry_msgs
https://github.com/Yadunund/test_ros2_pkgs

bai-ivan, 2023-05-09:
You are correct, probably i messed up something on the way. 
```
# Standard metadata for higher-level stamped data types.
# This is generally used to communicate timestamped data
# in a particular coordinate frame.

# Two-integer timestamp that is expressed as seconds and nanoseconds.
builtin_interfaces/Time stamp
        int32 sec
        uint32 nanosec

# Transform frame with which this data is associated.
string frame_id
```