osrf/ros2_test_cases #878: Sane development environment

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Test message generation with .msg, .srv, .action, and .idl on top of the installed environment**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Robotgir, 2023-05-10:
please assign this to me. Thankyou

Robotgir, 2023-05-10:
Hi i tested .msg, .srv and .action but how do i test .idl file, never came across such file before but understood after reading that it is a standard interface definition language, where all .msg .srv .action get converted to,  Please kindly point me in the right direction here

Robotgir, 2023-05-10:
@Yadunund  i have tested the generation of .msg, .srv, .action, but new to .idl, did some reading and understood that interface definition language  .idl format is what .msg, .srv and .action get converted to under the hood.

i came across the "rosidl generate package interface_file" command,
i am not able to find a example .idl to create one. can you please guide me in the right direction. Thankyou.

Yadunund, 2023-05-11:
Thanks for testing. I have a feeling the test cases should have been worded "Test message generation with .msg, .srv, .action **to** .idl on top of the installed environment". @clalancette does that make sense?

Robotgir, 2023-05-11:
@Yadunund i see that the .idl files are generated for all the .msg, .srv and .action files i created

clalancette, 2023-05-11:
> Thanks for testing. I have a feeling the test cases should have been worded "Test message generation with .msg, .srv, .action **to** .idl on top of the installed environment". @clalancette does that make sense?

So we support making messages from all of `.msg`, `.srv`, `.action`, and `.idl` files.  The pipeline basically goes something like:

`.msg` -> `.idl` -> generated Python/C++ code

So you can start with a `.msg` file, and end up with the correct generated code.  But you can *also* skip the `.msg` part and provide a `.idl` file directly (though we only support a subset of the full IDL specification).

Unfortunately we don't have a lot of good examples of IDL files.  One example we do have is here: https://github.com/ros2/rosbag2/tree/iron/rosbag2_test_msgdefs/msg .

So we shouldn't consider this test complete until we try out the `.idl` files as well.

Additionally, I think we should make an update to YATM to more explicitly spell out what we want to see done here.  I'll open a PR for that.

Robotgir, 2023-05-12:
I have created a sample TestMessage.idl inside "src/tutorial_interfaces/idl/TestMessage.idl" and used it in a publisher/subscriber example, and it works as expected. 
- can confirm that the c cpp py files are generated for all message types like .msg, .srv, .action and .idl
- find the attached images for results
[image]
[image]
[image]

Yadunund, 2023-05-16:
That's fantastic! Thank you for trying this out.

closed 2023-05-16 by Yadunund