osrf/ros2_test_cases #880: Service executables in `demo_nodes_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Add two ints**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>
- [ ] **Add two ints sync**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client_sync
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-07:
@clalancette @Yadunund could you please assign this issue to me? Thank you so much!

juliamp22, 2023-05-07:
**Add two ints** works perfectly!

[image]

juliamp22, 2023-05-07:
**Add two ints sync**: 

- gives an error that **No executable found** when executing _ros2 run demo_nodes_py add_two_ints_client_sync_
- changing  _ros2 run demo_nodes_py add_two_ints_client_sync_ to  _ros2 run demo_nodes_py add_two_ints_client_**async**_ which exists in *demo_nodes_py*, and worked perfectly:

[image]

juliamp22, 2023-05-07:
@clalancette @Yadunund this issue can be closed. Thank you so much!

clalancette, 2023-05-07:
Thank you for testing!

closed 2023-05-07 by clalancette