osrf/ros2_test_cases #881: Service executables in `demo_nodes_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Add two ints**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>
- [ ] **Add two ints sync**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client_sync
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

hwoithe, 2023-05-04:
I installed the Cyclone DDS packge
```
$ sudo apt install -y ros-iron-rmw-cyclonedds-cpp
```

Restarted the ros2 daemon

```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 daemon stop
$ ros2 daemon start
```

- [x] Add two ints

```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 run demo_nodes_py add_two_ints_client
service not available, waiting again...
[INFO] [1683236537.603174139] [add_two_ints_client]: Result of add_two_ints: 5
```

```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 run demo_nodes_py add_two_ints_server
[INFO] [1683236537.587631216] [add_two_ints_server]: Incoming request
a: 2 b: 3
```

- [ ] Add two ints sync

```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 run demo_nodes_py add_two_ints_client_sync
No executable found
```

Changing `add_two_ints_client_sync` to `add_two_ints_client_async` since [demo_nodes_py](https://github.com/ros2/demos/tree/iron/demo_nodes_py/demo_nodes_py/services) does not seem to have `add_two_ints_client_sync`.

- [x] Add two ints async

```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 run demo_nodes_py add_two_ints_client_async
service not available, waiting again...
[INFO] [1683236748.629480864] [add_two_ints_client]: Result of add_two_ints: 5
```

```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 run demo_nodes_py add_two_ints_server
[INFO] [1683236748.598894563] [add_two_ints_server]: Incoming request
a: 2 b: 3
```

Yadunund, 2023-05-05:
Thanks for testing! 

Indeed it should be `add_two_ints_client_async`. This has been corrected in the tool used to create this ticket https://github.com/audrow/yatm/pull/348 so future iterations will have the right command. We did not want to regenerate tickets with the correction so I've manually updated the description in this ticket instead.

Could you confirm if the test works when you use `add_two_ints_client_async`?

hwoithe, 2023-05-05:
Yep, the test for `add_two_ints_client_async` runs fine.

```
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 run demo_nodes_py add_two_ints_client_async
[INFO] [1683293533.832661883] [add_two_ints_client]: Result of add_two_ints: 5
```

```
$ env | grep RMW_IMPLEMENTATION
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 run demo_nodes_py add_two_ints_server
[INFO] [1683293533.819831462] [add_two_ints_server]: Incoming request
a: 2 b: 3
```

clalancette, 2023-05-05:
Fantastic, thank you for testing.  Given that we've already fixed the bug in the test generation tool, and the fact that the test itself is successful, I'm going to close this one out.

closed 2023-05-05 by clalancette