osrf/ros2_test_cases #883: Service executables in `demo_nodes_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Checks
- [ ] **Add two ints**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>
- [ ] **Add two ints sync**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client_sync
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-05:
I would like to take on this ticket.

Kartik9250, 2023-05-05:
## Checks
- [x] Add two ints
- [x] Add two ints sync
   
>   1. ```shell
>      # User input in terminal 1
>      ros2 run demo_nodes_py add_two_ints_client_sync
>      ```
There was no script `add_two_ints_client_sync` instead there was `add_two_ints_client_async`. So used that for testing, that worked fine. I know synchronous calls are thread block whereas asynchronous calls are non-thread blocking therefore I am not sure if that was the intended thing to test.

clalancette, 2023-05-06:
> There was no script add_two_ints_client_sync instead there was add_two_ints_client_async

Yeah, that was a typo in the test case.  We've already fixed it for next year.  Thank you for testing!

closed 2023-05-06 by clalancette