osrf/ros2_test_cases #897: Topic executables in `demo_nodes_py`

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener
       ```
  
    
  </details>
- [ ] **Check `listener` and `talker` QOS**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker_qos
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_qos
       ```
  
    
  </details>
- [ ] **Check serialized `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_serialized
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-06:
Hello, could you please assign this test to me?

juliamp22, 2023-05-06:
Check listener and talker works: 
[image]

juliamp22, 2023-05-06:
Listener and talker demo_nodes_py works: 

[image]

juliamp22, 2023-05-06:
Talker and listener serialized works:
[image]

juliamp22, 2023-05-06:
If you see all right, you can close the task @Yadunund. Thank you so much!

juliamp22, 2023-05-07:
@clalancette could you please assign this task to me? Lots of thanks!!

clalancette, 2023-05-07:
Thank you for testing!

closed 2023-05-07 by clalancette