osrf/ros2_test_cases #901: Topic monitor

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/master/topic_monitor/README.md)

## Checks
- [ ] **Comparing reliability QoS settings**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You will need two machines with ROS 2: one mobile and one stationary.
       
       
    1.
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display
       ```
    1.
       Run the `ros2 launch topic_monitor reliability_demo.launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.
       
       ```bash
       # User input in terminal 1
       ros2 launch topic_monitor reliability_demo.launch.py
       ```
    1.
       Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.
       
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display --allowed-latency 5
       ```
    1.
       Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
       
  
    **Expect**
    
    1.
       You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.
       
       [image]
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

juliamp22, 2023-05-06:
@Yadunund could you please assign this task to me? Thank you so much

juliamp22, 2023-05-07:
Appears the following error:
[image]

After executing ros2 launch topic_monitor reliability_demo.launch.py, the error says that 
_file 'reliability_demo.launch.py' was not found in the share directory of package 'topic_monitor' which is at 'C:\dev\ros2_iron\share\topic_monitor'_

The problem is that it is trying to search the launch file inside **\topic_monitor** and not inside **\topic_monitor\launch**. @Yadunund how I shoud proceed? Thank you so much.

juliamp22, 2023-05-07:
The monitor on a mobile machine such as a laptop - executing `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` Works but the topic_monitor windows does not respond.
[image]

Yadunund, 2023-05-08:
Thanks for testing! We updated our launch files for `iron` so the launch command should be `ros2 launch topic_monitor reliability_demo_launch.py`. I'll update the description here and in our future test cases.

juliamp22, 2023-05-09:
Hello @Yadunund , you are totally right, now I am able to execute in a stacionary machine the 
`ros2 run topic_monitor topic_monitor --display`
`ros2 launch topic_monitor reliability_demo_launch.py`

But the display does not display anything, stays (Not Responding)

juliamp22, 2023-05-09:
The same happens on a mobile machine:
[image]

juliamp22, 2023-05-09:
Hello @Yadunund how do you want to proced? Should I test anything else? Thank you so much.

Yadunund, 2023-05-10:
I see. Thanks for testing this again. I've logged this ticket to keep track of the issue  https://github.com/osrf/ros2_test_cases/discussions/1054.
Will close this ticket

closed 2023-05-10 by Yadunund