osrf/ros2_test_cases #902: Topic monitor

Yadunund, 2023-04-05:
## Setup
- DDS vendor: Cyclone
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/master/topic_monitor/README.md)

## Checks
- [ ] **Comparing reliability QoS settings**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You will need two machines with ROS 2: one mobile and one stationary.
       
       
    1.
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display
       ```
    1.
       Run the `ros2 launch topic_monitor reliability_demo.launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.
       
       ```bash
       # User input in terminal 1
       ros2 launch topic_monitor reliability_demo.launch.py
       ```
    1.
       Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.
       
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display --allowed-latency 5
       ```
    1.
       Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
       
  
    **Expect**
    
    1.
       You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.
       
       [image]
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Abishalini, 2023-05-03:
I'll test this

Abishalini, 2023-05-03:
I was able to see the results as mentioned above
[image]

The topic monitor did not shut down cleanly though
```
^CException in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/opt/ros/iron/lib/python3.10/site-packages/topic_monitor/scripts/topic_monitor.py", line 305, in run
    run_topic_listening(self.node, self.topic_monitor, self.options)
  File "/opt/ros/iron/lib/python3.10/site-packages/topic_monitor/scripts/topic_monitor.py", line 362, in run_topic_listening
    rclpy.spin_once(node, timeout_sec=0.05)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 208, in spin_once
    executor.spin_once(timeout_sec=timeout_sec)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 717, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 703, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 604, in _wait_for_ready_callbacks
    raise ExternalShutdownException()
rclpy.executors.ExternalShutdownException
Traceback (most recent call last):
  File "/opt/ros/iron/lib/python3.10/site-packages/topic_monitor/scripts/topic_monitor.py", line 430, in main
    time.sleep(time_to_sleep)
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/iron/lib/topic_monitor/topic_monitor", line 33, in <module>
    sys.exit(load_entry_point('topic-monitor==0.27.0', 'console_scripts', 'topic_monitor')())
  File "/opt/ros/iron/lib/python3.10/site-packages/topic_monitor/scripts/topic_monitor.py", line 434, in main
    data_receiving_thread.stop()
  File "/opt/ros/iron/lib/python3.10/site-packages/topic_monitor/scripts/topic_monitor.py", line 312, in stop
    rclpy.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown
    _shutdown(context=context)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:313
[ros2run]: Process exited with failure 1
```

Yadunund, 2023-05-04:
Thanks for testing!

Logged the shutdown issue here https://github.com/osrf/ros2_test_cases/discussions/1048

closed 2023-05-04 by Yadunund