osrf/ros2_test_cases #903: Topic monitor

Yadunund, 2023-04-05:
## Setup
- DDS vendor: FastDDS
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/master/topic_monitor/README.md)

## Checks
- [ ] **Comparing reliability QoS settings**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You will need two machines with ROS 2: one mobile and one stationary.
       
       
    1.
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display
       ```
    1.
       Run the `ros2 launch topic_monitor reliability_demo.launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.
       
       ```bash
       # User input in terminal 1
       ros2 launch topic_monitor reliability_demo.launch.py
       ```
    1.
       Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.
       
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display --allowed-latency 5
       ```
    1.
       Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
       
  
    **Expect**
    
    1.
       You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.
       
       [image]
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

bai-ivan, 2023-05-06:
working on this, but it will need a bit of time to install on stationary machine.

bai-ivan, 2023-05-06:
2) GUI hangs and crashes on both machines.
```
C:\Users\user>ros2 run topic_monitor topic_monitor --display
[ros2run]: Process exited with failure 3489660927
```
also this
```
C:\Users\user\test>ros2 launch topic_monitor reliability_demo.launch.py
file 'reliability_demo.launch.py' was not found in the share directory of package 'topic_monitor' which is at 'C:\dev\ros2-iron\share\topic_monitor'
```
can't proceed

bai-ivan, 2023-05-06:
After reinstall of everything (hoping to get it working) i managed to properly verify the test without the GUI. The test states 
```
[INFO] [1683409797.469751400] [topic_monitor]: /sensor_data_best_effort: 209
[INFO] [1683409798.600279700] [topic_monitor]: /sensor_data_best_effort: 211
[INFO] [1683409799.258277200] [topic_monitor]: ---------------
[INFO] [1683409799.656070000] [topic_monitor]: /critical_data: Stale
[INFO] [1683409800.234888700] [topic_monitor]: /sensor_data_best_effort: Alive
[INFO] [1683409800.668679800] [topic_monitor]: ---------------
[INFO] [1683409804.175167300] [topic_monitor]: ---------------
[INFO] [1683409804.465828100] [topic_monitor]: /critical_data: Stale
[INFO] [1683409804.467282200] [topic_monitor]: /sensor_data_best_effort: Stale
[INFO] [1683409804.957806400] [topic_monitor]: ---------------
[INFO] [1683409819.928637200] [topic_monitor]: /critical_data: 253
[INFO] [1683409819.932973700] [topic_monitor]: /sensor_data_best_effort: 253
[INFO] [1683409820.150405900] [topic_monitor]: ---------------
[INFO] [1683409820.156202200] [topic_monitor]: /critical_data: Alive
[INFO] [1683409820.161657500] [topic_monitor]: /sensor_data_best_effort: Alive
[INFO] [1683409820.170743800] [topic_monitor]: ---------------
[INFO] [1683409820.439961300] [topic_monitor]: /critical_data: 254
[INFO] [1683409820.446875600] [topic_monitor]: /sensor_data_best_effort: 254
[INFO] [1683409820.953307900] [topic_monitor]: /critical_data: 255
[INFO] [1683409820.959233500] [topic_monitor]: /sensor_data_best_effort: 255
```
after i leave the range of my router, but the GUI is not shown/frozen

Yadunund, 2023-05-08:
Thanks for testing.

The launch file name should be `reliability_demo_launch.py` which will be fixed for future testing. https://github.com/audrow/yatm/pull/359

I've also been encountering laggy performance of RQT GUIs on windows which may be related https://github.com/osrf/ros2_test_cases/issues/844#issuecomment-1537352558
cc: @clalancette 

I'll close this ticket as the basic functionality works.

closed 2023-05-08 by Yadunund