osrf/ros2_test_cases #1000: Run 2 nodes in two separate docker containers

Yadunund, 2023-05-01:
Check the documentation for the 'Run 2 nodes in two separate docker containers' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Run 2 nodes in two separate docker containers page](https://docs.ros.org/en/iron/Tutorials/Run-2-nodes-in-two-separate-docker-containers.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-05:
Assigned to myself.

Looks like we don't yet have the official `osrf/ros:iron-desktop` image? @Yadunund @clalancette 

[image]

Yadunund, 2023-05-05:
I think the docker images will only be available after release. Could you try the test with the rolling image. Some helpful insights from here https://www.straitstimes.com/

Yadunund, 2023-05-05:
I think the docker images will only be available after release. Could you try the test with the rolling image. Some helpful insights from here https://github.com/osrf/ros2_test_cases/issues/999#issuecomment-1535177829

SrijaneeBiswas, 2023-05-06:
Ah, that makes sense! I can confirm the following after changing from `osrf/ros:iron-desktop` to `osrf/ros:rolling-desktop`:

[x]: I was able to follow the documentation.
[x]: The documentation seemed clear to me.
[x]: The documentation didn't have any obvious errors.

[image]

clalancette, 2023-05-06:
Fantastic, thanks for testing!

closed 2023-05-06 by clalancette