osrf/ros2_test_cases #1001: Security Main

Yadunund, 2023-05-01:
Check the documentation for the 'Security Main' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Security Main page](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Security-Main.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-07:
Assigned to myself.

I can confirm that everything works as described in the documentation. On the **Examining Network Traffic** page -> **Display unencrypted data packets** section, I did not get the mentioned result for the following command:

`sudo tcpdump -i any -X udp portrange 7401-7500`

Instead, I got the following output:
```tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes

Let me know if I'm missing something.

clalancette, 2023-05-07:
Thank you for testing1

closed 2023-05-07 by clalancette