osrf/ros2_test_cases #1002: Setting up a Robot Simulation Ignition

Yadunund, 2023-05-01:
Check the documentation for the 'Setting up a Robot Simulation Ignition' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Setting up a Robot Simulation Ignition page](https://docs.ros.org/en/iron/Tutorials/Simulators/Ignition/Setting-up-a-Robot-Simulation-Ignition.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ayushghosh21, 2023-05-07:
I can take this!

ayushghosh21, 2023-05-07:
Probably not a big deal but after running ```ign topic -l```, I am getting additional lidar topics which are not shown in the example output.
```
ayushg@ayushg:~$ ign topic -l
/clock
/gazebo/resource_paths
/gui/camera/pose
/lidar
/lidar/points
/lidar2
/lidar2/points
/model/vehicle_blue/odometry
/model/vehicle_blue/tf
/stats
/world/visualize_lidar_world/clock
/world/visualize_lidar_world/dynamic_pose/info
/world/visualize_lidar_world/pose/info
/world/visualize_lidar_world/scene/deletion
/world/visualize_lidar_world/scene/info
/world/visualize_lidar_world/state
/world/visualize_lidar_world/stats

```

ayushghosh21, 2023-05-07:
Unable to find ros_gz_bridge package after following instructions in section 2.

```
ayushg@ayushg:~$ sudo apt-get install ros-iron-ros-ign-bridge
[sudo] password for ayushg: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-iron-ros-ign-bridge is already the newest version (0.244.3-3jammy.20230429.030614).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
ayushg@ayushg:~$ source /opt/ros/iron/setup.bash
ayushg@ayushg:~$ ros2 run ros_gz_bridge parameter_bridge /model/vehicle_blue/cmd_vel@geometry_msgs/msg/Twist]ignition.msgs.Twist
Package 'ros_gz_bridge' not found
ayushg@ayushg:~$ 

```

Yadunund, 2023-05-11:
Thanks for trying this. Turns out we're still in the process of releasing debians for [ros_gz](https://github.com/gazebosim/ros_gz) which is why you're unable to install it at the moment. Will update once that's ready.

clalancette, 2023-06-23:
This probably failed because we didn't have the debians available.

With the tutorial party over, I'm going to close this out.  Thanks for your efforts here.

closed 2023-06-23 by clalancette