osrf/ros2_test_cases #1003: Simulation Main

Yadunund, 2023-05-01:
Check the documentation for the 'Simulation Main' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Simulation Main page](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Simulation-Main.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-09:
Assigned to myself.

I was working on [Setting up a robot simulation (Gazebo)](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Gazebo.html) and had the following observations/errors:

- I couldn't get the simulation server and gui running with `ign gazebo -v 4 -r visualize_lidar.sdf`. Instead, I had to use `gz sim -v 4 -r visualize_lidar.sdf`.
- I had to replace all the `ros2 run ros_ign_bridge parameter_bridge` calls with `ros2 run ros_gz_bridge parameter_bridge`.
- After setting the bridge as above and publishing `ros2 topic pub /model/vehicle_blue/cmd_vel geometry_msgs/msg/Twist "linear: {x: 0.5}"` from another sourced terminal, I could not see the robot drive (I used a higher linear velocity just to check).
- I ran the parameter_bridge for the LaserScan data as well but could not visualize anything on RViz. Attached is a screenshot:
file:///home/dreamcatcher/Pictures/Screenshots/ign_sim_rviz_issue.png

Let me know what I'm missing and thanks for your help!

SrijaneeBiswas, 2023-05-09:
file:///home/dreamcatcher/Pictures/Screenshots/ign_sim_rviz_issue.png

Yadunund, 2023-05-11:
Thanks for trying this. Can I clarify if you were able to run `sudo apt-get install ros-iron-ros-ign-bridge`? From my understanding we haven't released these packages into `iron` yet and hence you should be seeing an error similar to https://github.com/osrf/ros2_test_cases/issues/1002#issuecomment-1537273255

SrijaneeBiswas, 2023-05-17:
Hi @Yadunund ,

Sorry for the delay in getting back. I could install `ros-iron-ros-ign-bridge` (I saw the packages were added a day after I ran my previous tests). So, I can now successfully run `ros2 run ros_ign_bridge parameter_bridge`.

However, the other three issues I mentioned on my comment still remain.

And, with the package name updated, do we want to update the documentation too? (The **Configuring ROS2** section still refers to it as `ros_gz_bridge`).

clalancette, 2023-06-23:
Thanks for the efforts here.  Since the tutorial party is over (and Iron is released), I'm going to close this out.  Any further problems can be tracked via the normal repository mechanisms.

closed 2023-06-23 by clalancette