osrf/ros2_test_cases #1004: Single Package Define And Use Interface

Yadunund, 2023-05-01:
Check the documentation for the 'Single Package Define And Use Interface' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Single Package Define And Use Interface page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Single-Package-Define-And-Use-Interface.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-05:
Taking this.

marioprats, 2023-05-05:
Works as expected.

clalancette, 2023-05-05:
Thanks for testing!

closed 2023-05-05 by clalancette