osrf/ros2_test_cases #1005: Sync Vs Async

Yadunund, 2023-05-01:
Check the documentation for the 'Sync Vs Async' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Sync Vs Async page](https://docs.ros.org/en/iron/Tutorials/Sync-Vs-Async.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

siddux, 2023-05-03:
Hi, I'd like to test this!

Yadunund, 2023-05-11:
@siddux any update?

siddux, 2023-05-11:
Sorry for the late response, I've performed all the tests without any errors and the documentation is clear. The results, depending if the service is running or not are the following:

[image]

[image]

clalancette, 2023-05-11:
Thank you for testing!

closed 2023-05-11 by clalancette